winginitau/VictronVEDirectArduino #8: Additional options for MPTT charge controller

teamkde, 2023-02-13:
I tried to create a branch so I could push this directly. If you want to add it these are confirmed to pull the additional values from an MPPT charger.

```
enum VE_DIRECT_DATA {
	VE_DUMP = 0,
	VE_SOC,
	VE_VOLTAGE,
	VE_POWER,
	VE_CURRENT,
	VE_ALARM,
 	VE_VPV,
	VE_PPV,
	VE_LAST_LABEL,
};

const char ved_labels[VE_LAST_LABEL][VED_MAX_LEBEL_SIZE] PROGMEM = {
		"Dump",	// a string that won't match any label
		"SOC",
		"V",
		"P",
		"I",
		"VPV",
		"PPV",
		"Alarm",
};
```

metrafonic, 2023-02-19:
Hey @teamkde i dont see your branch or pull request. Im thinking of doing the same thing for my smartsolar values

metrafonic, 2023-02-19:
@teamkde see this repo for complete mppt use https://github.com/RickardPettersson/VictronVEDirectArduino

metrafonic, 2023-02-27:
I ended up creating my own code which works wonders for me https://github.com/metrafonic/arduino-vedirect

davidrfloydii, 2023-03-02:
> I ended up creating my own code which works wonders for me https://github.com/metrafonic/arduino-vedirect

Does this work with SoftwareSerial or only hardware?

metrafonic, 2023-03-02:
> > I ended up creating my own code which works wonders for me https://github.com/metrafonic/arduino-vedirect
> 
> Does this work with SoftwareSerial or only hardware?

It should work fine with both, as long as softwareserial can handle `.readStringUntil(13);`